cuspaceflight/OkGo #20: fix booster

cillian64, 2015-07-19:
5V "booster" is not actually a 5v booster.  Also check it's up to 250mA / 1.5W

boyuanxiao, 2015-07-21:
Is this about the Murata chip?  It looks alright: http://uk.mouser.com/ProductDetail/Murata-Power-Solutions/MEJ2S0305SC/?qs=pitIP5Ybo%2F%252bvVODbaEhTwg%3D%3D (sorry if you've already checked this...)

But if the ground is the same on either side it could be cheaper to go for a non-isolated converter?

cillian64, 2015-07-21:
Yep, this issue was before I updated the part  -- previously I'd accidentally put on a 3v3 - 3v3 converter (which was also not up to the current of all the LEDs and the LCD backlight!).  I forgot to close the issue when I fixed that.

A non-isolated converter would work fine but I didn't find one as convenient on my brief search of farnell.  I am open to suggestions.

cillian64, 2015-07-21:
(fixed in d90080b024b719421ce510433d57c2e19026046a)

closed 2015-07-21 by cillian64

boyuanxiao, 2015-07-21:
Cool - looks like most step up regulators which are cheaper will require getting the right inductors etc. Probably not worth the hassle if you're only getting one.

cillian64, 2015-07-21:
Yep, cost is quite low down on my priorities, so it's all about the convenience of something that Just Works (and it helps that it's a pretty small footprint compared to a DIY switch-mode booster)